bootlin/training-materials #8: full-yocto-stm32-labs.pdf - Setup & Download Yocto section problems

l0cky3r, 2019-10-02:
Setup:
Go to $HOME/yocto-labs/ directory
Directory does not exist, we just created / were told that it is yocto-stm32-labs
I renamed it to yocto-labs...

Download Yocto:
As setup.

`git am $HOME/yocto-labs/bootlin-lab-data/0001-u-boot-update-config.patch
`
fails to apply:

```
~/yocto-stm32-labs/meta-st-stm32mp$ git am $HOME/yocto-stm32-labs/bootlin-lab-data/0001-u-boot-update-config.patch
Applying: u-boot: update config
.git/rebase-apply/patch:57: trailing whitespace.
-- 
error: patch failed: recipes-bsp/u-boot/u-boot-stm32mp-common_2018.11.inc:24
error: recipes-bsp/u-boot/u-boot-stm32mp-common_2018.11.inc: patch does not apply
.git/rebase-apply/patch:59: new blank line at EOF.
+
Patch failed at 0001 u-boot: update config
Use 'git am --show-current-patch' to see the failed patch
When you have resolved this problem, run "git am --continue".
If you prefer to skip this patch, run "git am --skip" instead.
To restore the original branch and stop patching, run "git am --abort".
```

michaelopdenacker, 2019-10-02:
Hi,
Thank you very much for the bug report. I believe we've found it too.
Stay tuned. We will push fixes in the next days.
Cheers,
Michael.

alexandrebelloni, 2022-08-24:
Fixed by 6677f99111f6bc3ac6562d26610fcd149c4ce9e3

closed 2022-08-24 by alexandrebelloni