bootlin/training-materials #64: cannot compile u-boot for vexpress_ca9x4_defconfig

Subhankar2000, 2021-12-02:
I have followed all the instructions as per this slides : [https://bootlin.com/doc/training/embedded-linux-qemu/embedded-linux-qemu-labs.pdf](url) to build the cross-compiler, but after that while building u-boot, I am getting this error, can anyone help me share the cross-compiler .config file such that I can tally, as I think the problem is with compiler only, but I have followed all the instructions while building  compiler.
Here is my u-boot building error, I have also followed all steps for making the u-boot, but it is not working !
`{standard input}: Assembler messages:
{standard input}:43: Error: selected processor does not support `dsb sy' in ARM mode
{standard input}:47: Error: selected processor does not support `isb sy' in ARM mode
{standard input}:240: Error: selected processor does not support `dsb sy' in ARM mode
{standard input}:244: Error: selected processor does not support `isb sy' in ARM mode
{standard input}:367: Error: selected processor does not support `dsb sy' in ARM mode
{standard input}:459: Error: selected processor does not support `dsb sy' in ARM mode
{standard input}:463: Error: selected processor does not support `isb sy' in ARM mode
{standard input}:596: Error: selected processor does not support `dsb sy' in ARM mode
make[1]: *** [scripts/Makefile.build:279: arch/arm/cpu/armv7/cache_v7.o] Error 1
make: *** [Makefile:1727: arch/arm/cpu/armv7] Error 2
`

michaelopdenacker, 2021-12-02:
Greetings
To confirm the cross-compiler issue, you could also try the cross-compiler provided by the distribution.
sudo apt install gcc-arm-linux-gnueabi
export CROSS_COMPILE=arm-linux-gnueabi-
If this works, there is an issue indeed with your cross-compiler. Otherwise, you may share your U-Boot .config file here.
I don't have a .config file for crosstool-ng for the QEMU lab at hand though (would take time to reproduce, and we only have time to support the participants to our sessions).
Cheers
Michael.

michaelopdenacker, 2021-12-02:
Closing this bug. No proof there is an issue in our training materials.
Don't hesitate to continue to post comments anyway.

closed 2021-12-02 by michaelopdenacker

Subhankar2000, 2021-12-02:
Thank You very much for the reply, I tried with the already available compiler but still the same error. I managed to get a proper compiler using buildroot ( only making the sdk ) and got everything working, however I am facing a new issue for nfs-krenel-server, I did everything according to the material to get the root file system from u-boot using the nfs-kernel-server, but in qemu it is stuck at starting kernel and in my host system where the server is, I am getting this in my logs, checked internet, many kind of solutions but none of them worked.
```
authenticated mount request from
```
that's it, I can't go further. I am on Ubuntu 20.04.3