bootlin/training-materials #129: Broken labs link in the presentation

gmsotavio, 2022-09-12:
Hello,

I was looking for the lab docs for BBB in the Embedded Linux training slide but apparently this link is broken.

https://github.com/bootlin/training-materials/blob/a97ebe4307585618ecadb0e5cec35d5e1f9cf8e2/slides/beagleboneblack-board-also-supported/beagleboneblack-board-also-supported.tex#L9

michaelopdenacker, 2022-10-04:
My fault here. The link was not supposed to exist.
We updated the materials to say that the instructions for the BBB board will be available by the end of the year (busy days!)
https://github.com/bootlin/training-materials/commit/028518318804dd7a69c22e432d9b4c667ad4ad9f
Thank you for reporting this issue!

closed 2022-10-04 by michaelopdenacker

michaelopdenacker, 2023-05-02:
Oops, this issue is fixed as we now have instructions for the BBB and BBBW.
See https://bootlin.com/doc/training/embedded-linux/embedded-linux-slides.pdf#page=12
and https://bootlin.com/doc/training/embedded-linux-bbb/

Thanks
Michael.