bootlin/training-materials #136: embedded-linux: build the Linux kernel with Buildroot

tpetazzoni, 2022-10-14:
The trick of adding the kernel modules to the rootfs-overlay is really ugly and can easily confuse people. In addition, in one of the next lab, we do build the Linux kernel with Buildroot, but do not actually use it, as we need the Nunchuk out of tree kernel driver.

This doesn't feel very pretty to present. I would suggest that we build the Linux kernel with Buildroot from the first Buildroot lab, so we have a clean setup. Remains to see how we build the Nunchuk driver. Perhaps provide a pre-defined package that people just have to copy to Buildroot's package/ directory ?

michaelopdenacker, 2022-10-14:
I agree that's very ugly, and that a solution wasn't straightforward to find.
OK to this in a cleaner way by adding a pre-defined package to compile the out of tree Nunchuk driver.

michaelopdenacker, 2022-11-02:
Fixed by https://github.com/bootlin/training-materials/commit/3076c225b063ec22c8447f183ad5e16bcdc102f7

closed 2022-11-02 by michaelopdenacker