bootlin/training-materials #139: yocto: properly describe the sysroot tasks

lucaceresoli, 2022-11-22:
In slide "Dependencies 1/2" we have this:

* DEPENDS = "recipe-b": the local do_prepare_recipe_sysroot task depends on the do_populate_sysroot task of recipe-b.
* RDEPENDS:${PN} = "package-b": the local do_build task depends on the do_package_write_<archive-format> task of recipe b.

Mentioning the do_*sysroot tasks here is confusing. We are still in the "Writing recipes - basics" section, which is pretty introductory and the per-recipe sysroot has not been mentioned yet. And later on we don't cover the topic at all.

I think we should:
1. remove the mention of do_*sysroot in the basics section (DEPENDS and RDEPENDS can be explained via common sense at this stage)
2. add an explanation of the per-recipe sysroot later on (in "Writing recipes - going further"?)
3. add a picture similar to the one in slide "Package splitting" to show the logical chain between the final tasks of a recipe and the preliminary tasks of a recipe that DEPENDS on it
4.  possibly also have a slide that explains how RDEPENDS ends up in writing a dependency into the generated RPM/DPKG/IPKG; this would probably halp demistifying why we need ":${PN}" after RDEPENDS and not after DEPENDS

Another possibility is to just stop at point 1 and not mention at all the topic.

alexandrebelloni, 2022-11-22:
I need do_prepare_recipe_sysroot because I usually explain that it is not worth generating the dependency graph and instead you can grep in the dot file. For example: grep busybox.do_prepare_recipe_sysroot task-depends.dot

lucaceresoli, 2022-11-22:
I see Alex, make sense. OTOH I have a hard time in trying to explain that slide.

But if I know we have a good explanation later on I'd be fine in just telling "this is advanced, explained later, but you can grep for these strings to follow dependencies instead of generating a graph".

Does that seems a good plan?

lucaceresoli, 2023-04-12:
Just committed.

closed 2023-04-12 by lucaceresoli