bootlin/training-materials #175: Embedded Linux: CONFIG_PERF_EVENTS must be enabled

adrian-martin, 2023-02-22:
In the *appdev* lab, the `perf` command doesn't work because the PERF_EVENTS were not enabled in the kernel's configuration. Everything works fine when using a kernel compiled with *CONFIG_PERF_EVENTS=y*

michaelopdenacker, 2023-02-22:
Hi  Adrian
Many thanks for the bug report!
Which board are you using?
Cheers
Michael.

adrian-martin, 2023-02-22:
Hi Michael,

I'm using a *SAMA5D2 xplained* board. I have noticed that *CONFIG_PERF_EVENTS* is enabled by default when using *multi_v7_defconfig* as configuration target, but it doesn't when using either *stm32_defconfig* or *sama5_defconfig* (that's why I had to mark it explicitly in the last lab). I think that I misunderstood the sentence "By running make help, look for the proper Makefile target to configure the kernel for your processor." in the section *Linux kernel configuration* of the *Kernel - Cross-compiling* lab

michaelopdenacker, 2023-03-28:
Thanks. Indeed, we cannot add all necessary settings in all possible configurations. The instructions would be too complicated.
Thanks anyway!