bootlin/training-materials #179: debugging: add a slide about smem tool which generates memory usage

clementleger, 2023-03-21:
https://www.selenic.com/smem/

tpetazzoni, 2023-03-21:
Don't know if it is that important, but smem has not seen any release since 2013, and the last commit in the Mercurial repo is from 2015. Perhaps it's the kind of tool that doesn't really need any maintenance: it just works and does its job. But I thought it was worth pointing that out.

Tropicao, 2025-01-03:
I agree on this, I would have a very hard time recommending the tool, since even the documentation seems deprecated now (eg: requesting python2 as a dependency). The only time I mentioned this one was when searching for a tool able to directly output the USS value of a process, but IMHO we should rather give hints about how to retrieve this value from `/proc/<pid>/smaps`

closed 2025-01-03 by Tropicao