bootlin/training-materials #180: debugging: add a slide about OpenOCD

clementleger, 2023-03-21:
OpenOCD is really poweful when debugging low-level kernel, maybe a slide and a lab might be really useful for kernel debugging.

tpetazzoni, 2023-03-21:
We have a very short single slide on JTAG/OpenOCD in the kernel training, but it's definitely very short. When it comes to doing a lab, we should not forget about the impact on on-site training: if a separate JTAG probe is needed, then it should be planned as part of the HW to have available for the course. Of course if the JTAG interface is built into the board through some FTDI-like interface, it's totally fine as it would not require extra hardware.

clementleger, 2023-03-21:
The ST-LINK debug itnerface is embeeded on the board wghich makes it an ideal candidate for JTAG debugging actually :) No need for any additional hardware

tpetazzoni, 2023-03-21:
Indeed, sounds perfect!

Tropicao, 2025-06-07:
- [ ] on slides part:
  - [ ] mention available technologies (JTAG/SWD)
  - [ ] introduce OpenOCD
  - [ ] explain basic OpenOCD usage
- [ ] in labs: add a basic OpenOCD use case. It may be integrated very close to the KGDB part
  - [ ] even nicer: it could be used to cover a case that is not doable with KGDB (eg: a bug occuring earlier than KGDB support being ready)