bootlin/training-materials #208: debugging: explain how to configure debugging in GUI (IDE)

Tropicao, 2023-07-24:
Trainees from different sessions has mentioned that it would be very convenient for them to be able to use the debugger directly in their IDE (like some of them are used to with their current tools, for bare metal for example)

We could either:
- deliver a configuration sample for 1 IDE in the userspace debugging lab data
- add a slide to show an example on how to configure it for 1 IDE

Since each trainee can have its own IDE, it is up to them to configure their IDE, but quick tests has shown it is pretty straightforward to configure debugging in VSCode, it may be a very good candidate to show a quick example. It should probably remain a "bonus" and not become a foreground topic, since the training is about learning about gdb usage

tpetazzoni, 2023-07-24:
Note: we already have a demo of VSCode + gdbserver for remote debugging in the sysdev course. Maybe some inspiration can be taken from there. However, the setup of the lab in the sysdev course is not very nice, with some home-grown scripts, a pre-defined VSCode project with several hard-coded paths. It works, but it doesn't look very pretty.

Tropicao, 2023-07-24:
Great, I'll definitely take a look at it and see if something can be shared and/or improved :)

commit 36dd8775 (2024-01-10): labs: debugging: add IDE debug configuration

Fixes #208
commit 944bf0a1 (2024-01-10): debugging: labs: add VSCode configuration for bonus lab

Fixes #208
closed 2024-01-10 by Tropicao