bootlin/training-materials #212: Misc typos in the Yocto course

tpetazzoni, 2023-10-11:
These typos/suggestions have been made by Robert PJ Day.

25: Both boxes, I would not say "from http, ...", I would write "via http, ..."
29: "that allows [one] to build"
29: "It has been initiated" -> "It was established"
32: "bitbake" should be rendered in monospaced font everywhere it is referred to as the program name; if you are simply referring to the generic bitbake technology, use "BitBake".
33: "Tasks have dependencies over other tasks" -> "Tasks might have build-time dependencies on other tasks"
34: "composed together" is redundant; just say "processed"
35: "The word Poky (pronounced Pock-ee) has several meanings"
38: Rather than saying "Editing Poky or meta-ti is a no-go", why not be more general, "Editing Poky or any vendor layer such as meta-ti is a no-go"?
40: "Might be worth mentioning that kirkstone is an LTS release; otherwise, readers might wonder why you're selecting that branch
55: I would introduce site.conf, for across-project settings such as local source mirrors, that sort of thing. I particularly use site.conf in every project for just that purpose.
62: "A Recipe describe[s] ..."
64: "overridden or modified" is kind of redundant
72: I would introduce "bitbake -e" so students can verify what the final settings are.
73: "Package variants"?  (singular?)
74: Not all variants are "virtual". For example, both systemd and eudev provide "udev".
75: I would explain "PROVIDES =" at this point, since it's so closely related to what is on this page.
77: Is it true that a "_git" recipe version is considered the highest available version? If so, you should mention that.
80: You sure PACKAGE_EXCLUDE does not work with .deb packages? meta/lib/oe/package_manager/deb/__init__.py. I tried it, it seems to work.
80: Probably want to render RDEPENDS in monospaced font
82: "optional options" -> "additional options"
84: Why not just explain that the name of the dir is "sstate-cache"?
84: "This cache is use[d] ..."
89: I would emphasize that a single recipe can generate numerous packages, and even list some of them -- dev, doc, libs, and so on
91: Might explain that the name "PN" is an unfortunate holdover, given that these days it really refers to the recipe name, not a package name
99: "when a file is found in a paths..." ????
106: "Sometimes a recipe [has] dependencies ..."
107: "Default tasks already exist[s] ..."
111: I might mention that a lot of "addtask" is in bbclass files, where specialized builds need to define additional tasks; good example is kernel.bbclass, which defines tasks that make sense only for building kernels
113: "To apply patches before they [make] their way ..."
114: Patch files are also recognized automatically if ending in .gz, .bz2, etc. See meta/lib/oe/patch.py.
128: I believe that wildcarding with bbappend can wildcard only the right-hand side of the version -- that is, the "%" character can appear only just before the ".bbappend" suffix, no? If true, this is worth explaining.
132: "PROVIDES =" can be used not just for virtuals; both systemd and eudev contain the line "PROVIDES = udev". I mentioned this earlier. Not sure where you would mention that "PROVIDES =" is used extensively to "provide" alternate names for a recipe
134: I would point out you can inherit multiple class files at the same time
135: These days, kernel.bbclass (and others) are under meta/classes-recipe. And there's meta/classes-global as well.
136: Some of those tasks have been moved to classes-global/utility-tasks.bbclass
142: "RPM/DPKG" -> "RPM/DEB"??
144: If it's a .conf file, don't you need to use "INHERIT +="? Not "inherit"? The very next slide says just recipes and classes.
152: mirrors.bbclass is now in classes-global
154: I think this will generate tarballs of *all* downloaded SRC_URIs, not just git repos.
159: I would explain that what makes something a "layer" is the existence of a conf/layer.conf file, and that's why "poky" itself is *not* a layer, but (as you say) incorporates a number of actual layers. If you want to mention another example, the "meta-ti" checkout is not a layer, but it includes two layers.
164: Again, I would be careful about calling things like meta-ti a "layer".
175: I would mention meta-yocto-bsp and meta-ti-bsp as examples of BSP layers.
180: bluetooth is not a MACHINE_FEATURES, it's a DISTRO_FEATURES, the way it's
     used by connman. You might want to introduce both of those variables as
     they are both possibly consulted by machine configuration files. Also
     TUNE_FEATURES?
216: IIRC, the only difference between and .wks and a .wks.in file is that
     the latter allows variables to be expanded
219: Packagegroups are not always found under recipes-core
220: I would specifically discuss packagegroup-base, which explains how 
a lot  of the initial configuration happens

lucaceresoli, 2023-10-13:
In progress. Committed most of the suggested improvements up to slide 99.

lucaceresoli, 2023-10-27:
Done up to slide 128

lucaceresoli, 2023-10-31:
Done up to slide 180

lucaceresoli, 2023-10-31:
All done.

closed 2023-10-31 by lucaceresoli