bootlin/training-materials #238: Mention the "-r" option of bitbake-getvar before lab #2

cperrysfl, 2024-04-27:
Hello!

bitbake-getvar is a great tool to understand what's going on when adding dropbear to the image from local.conf. It allows the trainee to see variable assignation in action rather than going off the theory.

There is currently a slide on bitbake-getvar before lab#2 (73/321) but that slide does not mention the "-r" option which is needed to debug IMAGE_INSTALL for core-image-minimal. Otherwise, bitbake-getvar uses only the configuration metadata and things look fine.

Note that there is mention of the "-r" option on slide 124/321 but that comes after lab#2.

lucaceresoli, 2024-04-29:
Hi @cperrysfl,

thanks for your comment.

We don't mention `-r` in slide 73 because we haven't yet introduced recipes at that point. Of course building the first image in lab 2 involves using a recipe: unavoidably some topics need to be mentioned somehow before they are actually explained, even though we have put a lot of effort in minimizing such situations. Trainers are free to quickly mention `-r` while debugging in lab 2 if so needed, which is what I do at times. However I think the logical flow of the slides should be kept to avoid confusion.

Closing this, but feel free to reopen if you have more arguments that I'm missing.

closed 2024-04-29 by lucaceresoli

cperrysfl, 2024-04-29:
Ok, good point.

Thank you!