bootlin/training-materials #258: Improve probes introduction

Tropicao, 2024-12-10:
Trainees are often confused about kprobes/uprobes and their differences with other tracing solutions. The materials could be improved with the following adjustments:
- have 1 generic slide about probes solutions offered by the kernel (uprobes, fprobes, kprobes) , explaining the benefits of probes (instrumenting code without having to rebuild it), but also their downsides. This slide would be just before the first probe mention, so in the application tracing part, before introducing uprobes
- remove some low levels details that will likely never be used by trainees:
  - kernel APIs for kprobes
  - kernel APIs for kretprobes
- replace the slides mentioned above with some examples about how to manipulate uprobes and kprobes from tracefs, which is far more useful
  - how to insert a uprobe from /sys/kernel/tracing/uprobe_events in application tracing part
  - how to insert a kprobe from /sys/kernel/tracing/kprobe_events in system wide tracing part

While perf should still be the preferred solution to manipulate probes, trainees would learn about the lowest level interface offered to user and know how to handle probes if no intermediate tool is available on their platform

Tropicao, 2024-12-10:
@lucaceresoli I am pretty interested in your opinion on this matter :)

lucaceresoli, 2024-12-17:
I fully second you proposal.

Only I think it makes sense to keep a reference to the kernel APIs for k(ret)probes, but reduced to a mere bullet, to give the big picture. E.g. "kprobes and kretprobes can be registered by kernel code calling register_kprobe() and register_kretprobe()".

closed 2025-01-02 by Tropicao